CentroEPiaggio/kuka-lwr #15: OneTaskInverseKinematics produces speed limit

Tabjones, 2015-03-11:
Launching lwr_launch with controller=OneTaskInverseKinematics and stopped_controllers=joint_position_trajectory_control produces a speed limit J1 from the kuka.
This happens as soon as handshake is done and controller is loaded, with no actual command given.
Could be related to commit 5b57426d33b673a36a1eb35dca28d5813368f076

Tabjones, 2015-03-11:
Issue is not affecting d789bfdbb41dabc885579f2d6af70b5fc75b6255

carlosjoserg, 2015-03-11:
You mean that the error is not present in d789bfd ?

Tabjones, 2015-03-11:
yes, this error is not present in d789bfd

carlosjoserg, 2015-03-12:
I merged because we are using it in the `multi-robot-test` branch like this, but it is true that we are using the position controllers. 

Could you try setting the initial joint stiffness to zero at [this line](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_hw/src/lwr_hw.cpp#L111)?

Tabjones, 2015-03-12:
i will try as soon as i can.

carlosjoserg, 2015-06-25:
@Tabjones We have several updates ahead from this issue that I believe solve it, so I will close it. If it happens again, don't hesitate to re-open. Thanks for reporting !

closed 2015-06-25 by carlosjoserg