CentroEPiaggio/kuka-lwr #17: Out of the box readme example not working

gpldecha, 2015-04-23:
Hi have tried the lasted version and when I run: $ roslaunch lwr_launch lwr.launch
I get the following error:

[ERROR] [1429780242.515969375, 0.494000000]: Could not find joint 'lwr_0_joint' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1429780242.516077306, 0.494000000]: Failed to initialize the controller
[ERROR] [1429780242.516109259, 0.494000000]: Initializing controller 'joint_position_trajectory_controller' failed

carlosjoserg, 2015-04-23:
Hi, very sorry for that.

We are planning an important update soon merging the `multi-robot-test` branch where everything is in a much more stable state.

Nevertheless, that problem is because the interface for the joints is set to [EffortJointInterface](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_description/model/kuka_lwr.transmission.xacro#L11), and the default controllers are supposed to work in a `PositionJointInterface`. You can change that and it will work. Currently there is no support in `ros_control` for multiple hardware interfaces, and you need to witch them by hand.

However, I would suggest working with the `multi-robot-test` branch if you are in a hurry.

Let me know if it works.

carlosjoserg, 2015-06-25:
@gpldecha I'm closing this. If you have troubles again just let us know. Thanks for reporting.

closed 2015-06-25 by carlosjoserg