CentroEPiaggio/kuka-lwr #27: lwr_controllers: 'kdl' is not a valid build or run depends

gavanderhoorn, 2015-06-12:
The package manifest of the `lwr_controllers` pkg lists `kdl` as a `build_depend` and a `run_depend` ([here](https://github.com/CentroEPiaggio/kuka-lwr/blob/bb1b9eeea368213b2b9c737f4d454ca04293d6af/lwr_controllers/package.xml#L23)). `rosdep` cannot resolve that key, and I can also not find it in the online [rosdistro/rosdep/base.yaml](https://github.com/ros/rosdistro/blob/5c8b7d54406aec4f7ff6bf691df8081a5610dd29/rosdep/base.yaml) file.

carlosjoserg, 2015-06-16:
Thanks. Actually in the CMakeLists.txt is listed as a non-catkin package, and twice, [here](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_controllers/CMakeLists.txt#L4) and [here](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_controllers/CMakeLists.txt#L23).

But I'm actually linking to the one delivered with ros, that is, [this one](http://wiki.ros.org/orocos_kdl).

So, I'm gonna use `<build_depend>orocos_kdl</build_depend>` and `<run_depend>orocos_kdl</run_depend>`, and in the CMakeLists.txt I'll use it as an actual catkin component like 
``` cmake
find_package(catkin REQUIRED COMPONENTS
  controller_interface
  control_msgs
  forward_command_controller
  control_toolbox
  realtime_tools
  urdf
  roscpp
  kdl_parser
  message_generation
  cmake_modules
  orocos_kdl
)

catkin_package(
#  DEPENDS 
  CATKIN_DEPENDS
    controller_interface
    control_msgs
    control_toolbox
    realtime_tools
    urdf
    forward_command_controller
    kdl_parser
    message_runtime
    orocos_kdl
  INCLUDE_DIRS include
  LIBRARIES ${PROJECT_NAME}
)
```

It works for me, I'll be pushing this change in while unless you see other inconveniences.

nrontsis, 2015-06-17:
After the changes you introduced in the CMakeLists.txt (mentioned above), I have trouble compiling:

```
CMake Error at /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:176 (message):
  catkin_package() CATKIN_DEPENDS on 'orocos_kdl', which has been found in
  '/opt/ros/indigo/share/orocos_kdl/orocos_kdl-config.cmake', but it is not a
  catkin package
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/catkin_package.cmake:98 (_catkin_package)
  kuka-lwr/lwr_controllers/CMakeLists.txt:40 (catkin_package)

-- Configuring incomplete, errors occurred!
```

I am too linking to http://wiki.ros.org/orocos_kdl

carlosjoserg, 2015-06-17:
Thanks, with all tests, I pushed the wrong version !

It is fixed e78ec82c5ead9979cb11530ff212e482e495819d

nrontsis, 2015-06-17:
Thanks :+1:

nrontsis, 2015-06-17:
Actually, I think that there might be still an issue.

When compiling to a **new** workspace errors pop up.

However, with the following change, it seems there is no issue:

```
find_package(catkin REQUIRED COMPONENTS
  controller_interface
  control_msgs
  forward_command_controller
  control_toolbox
  realtime_tools
  urdf
  roscpp
  kdl_parser
  message_generation
  cmake_modules
)

find_package(orocos_kdl REQUIRED)
```

carlosjoserg, 2015-06-17:
Oh, the old __new__ workspace technique always reveals errors, this is where travis will be of great help.

However, if you type `roscd orocos_kdl` it will take you to `/opt/ros/indigo/share/orocos_kdl`, where there is a `package.xml` defining the catkin package and cmake config files. I think the problem is where @gavanderhoorn pointed it above. I issued it as a question in ros/rosdistro#8702

carlosjoserg, 2015-06-18:
Ok, I just realized that kdl_parser already depends on orocos_kdl. So, there is no need to use `find_package(orocos_kdl REQUIRED)`.

In a new catkin_ws, 1242b8a1f138424ce287ed3b48a1c0b3f07750ba works for me, links to orocos-kdl, and so on.

marcoesposito1988, 2015-06-25:
It also compiles for me in a new workspace.

@nrontsis: can we close this?

nrontsis, 2015-06-25:
Sure!

marcoesposito1988, 2015-06-25:
Very good, thanks! @carlosjoserg: we can kill this issue.

carlosjoserg, 2015-06-25:
Good, thanks.

commit 5454b5cb (2015-06-16): orocos-kdl is the package is used here, this fixes #27
closed 2015-06-25 by carlosjoserg