CentroEPiaggio/kuka-lwr #29: Running on gazebo < 4

nrontsis, 2015-06-17:
I am trying to integrate kuka-lwr with SHADOW hand.

I see that in your requirements you state "gazebo 4 or higher".

What are the problems on running on gazebo 2.2 (the default for indigo)?

Thank you!

Nikitas

carlosjoserg, 2015-06-17:
The main difference is how to access the joint angles/positions [here](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_hw/src/default_lwr_hw_sim.cpp#L273) in API terms. It could be easily solved using a macro like [this one](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/indigo-devel/gazebo_ros_control/src/default_robot_hw_sim.cpp#L299-L303).

The reason why I didn't do it is basically to promote the use of newer versions of gazebo, since the install procedure is not so tedious.

nrontsis, 2015-06-17:
Okay thank you a lot. I will come back later to you, to close the issue or comment further.

Cheers!

nrontsis, 2015-06-25:
After experimenting a bit the package, I think that there is currently no issue with using Gazebo 2.2.1. It compiles, and runs out of the box. [I definitely do not like to use outdated versions, but unfortunately, in the project I work we have decided on ros-indigo and gazebo 2.2.1 :/ ]

In contrast, vito robot and the kuka version that it used there https://github.com/CentroEPiaggio/kuka-lwr/commit/b91e1944e3eaa3ac67c4664b4cff1e55c1a237af requires Gazebo 4, due to the reasons @carlosjoserg mentioned before (and especially due to Joint::SetStiffness() to which I couldn't find any alternative available in Gazebo 2.2.1).

I am closing the issue, but feel free to comment to what I wrote. :)

carlosjoserg, 2015-06-25:
Thanks.

However, it is strange that it compiles and run out of the box, it shouldn't, are you sure? I see in your timeline you are actually not using the lwr gazebo plugin [here](https://github.com/nrontsis/kuka-lwr/commit/bcc23e58724b82152abdc3cf221a6b0cbe715400), and in fact, this is why the robot falls in simulation without controllers as you mentioned in https://github.com/shadow-robot/sr-ros-interface/issues/333.

The `Joint::SetStiffness()` is not in the kuka package, but in the Pisa/IIT SoftHand, as an attempt to introduce springs in joints, but didn't work for me. I sent you to Vito to show you how I implemented a custom transmission for the hand that is compatible with ros control as an example to implement shadow robot transmissions, and how everything is assembled using the kuka as well.

nrontsis, 2015-06-26:
@carlosjoserg according to the code you sent me, for the difference in API terms of gazebo 4 vs gazebo 2:
```
#if GAZEBO_MAJOR_VERSION >= 4
        sim_joints_[j]->SetPosition(0, joint_position_command_[j]);
#else
        sim_joints_[j]->SetAngle(0, joint_position_command_[j]);
#endif
```
Hence gazebo 2.2.1 should replace `SetPosition` to `SetAngle`. However you **do not use** at all the `SetPosition` function in kuka-lwr package.

I tested again using kuka-lwr package (**untouched**) in Gazebo 2.2.1, and I did not find any issue.

You are right however that my issues for Vito robot in Gazebo 2.2.1 came from Pisa/IIT SoftHand. And vito example indeed helped me in connecting shadow with kuka.

carlosjoserg, 2015-06-26:
> However you do not use at all the SetPosition function in kuka-lwr package.

You are absolutely right, I was thinking in the hand, hence vito.

However, I plan to use the `SetPosition` next week, if it is only one place that I have to do that, no problem to add it, if there are more places, I would prefer not. 

So, yes, thanks for re-opening the issue, keep it that way until that is done.

nrontsis, 2015-06-26:
Okay, nice :blush:

In case where you use extensively the `SetPosition` I can easily put macros to handle it in a branch of my fork.

[Btw, I removed the lwr gazebo plugin in my branch because it was conflicting with the sr_gazebo_plugins used in SHADOW hand. I will ask the SHADOW guys [@ugocupcic] about it and maybe I will come back to you with a few more questions :innocent:]

Cheers!

ugocupcic, 2015-06-26:
@nrontsis we have some plans to get rid of that sr_gazebo_plugins package but need to look into it a bit more as it's not trivial.

What has been done in other projects is to spawn the 2 robot models separately in gazebo (one for the arm, one for the hand). But then it leads to quite a few nightmares (need to run some joint states merger etc...)

nrontsis, 2015-06-26:
@ugocupcic @carlosjoserg  I think it is better to continue the discussion about kuka+shadow in https://github.com/shadow-robot/sr-ros-interface/issues/333

carlosjoserg, 2015-07-27:
@nrontsis I added the check for the version in the gazebo interface [here](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_hw/include/lwr_hw/lwr_hw_gazebo.hpp#L81-L85), could you confirm it works for you and close the issue if so? Thanks.

gpldecha, 2015-12-17:
I have installed gazebo6 and when I launch the simulator I get: 

[WARN] [WallTime: 1450348091.185336] [14.746000] Controller Spawner couldn't find the expected controller_manager ROS interface.

carlosjoserg, 2015-12-17:
Could deveop more the question? How are you launching? Are you using fully the example given [here](https://github.com/CentroEPiaggio/kuka-lwr/tree/master/single_lwr_example) or your own?

gpldecha, 2015-12-17:
Sorry about my hast no issue. I made a mistake.

carlosjoserg, 2015-12-17:
No worries.

Anyway, looked like the lwr plugin was not loaded... might be things like
gazebo is started with pause on or the tag not being converted to SDF...

Let us know if it is gazebo6 compatible ;)

gpldecha, 2015-12-18:
It is indeed gazebo compatible.

carlosjoserg, 2015-12-18:
:+1: thanks !

closed 2016-10-10 by carlosjoserg