CentroEPiaggio/kuka-lwr #32: Compilation error

nrontsis, 2015-06-19:
I am getting the following error when running `catkin_make`:

```
kuka-lwr/lwr_controllers/src/joint_impedance_controller.cpp:94:101: error:
no ‘void lwr_controllers::JointImpedanceController::commandConfiguration(const ConstPtr&)’
member function declared in class ‘lwr_controllers::JointImpedanceController’
void JointImpedanceController::commandConfiguration(const std_msgs::Float64MultiArray::ConstPtr &msg){
```

@marcoesposito1988 you renamed `JointImpedanceController::command` to `JointImpedanceController::commandConfiguration` in https://github.com/CentroEPiaggio/kuka-lwr/commit/7319f24d487e67862acc40a905540018092f01c6 without changing the header file

marcoesposito1988, 2015-06-19:
@nrontsis: you're right, sorry about that. It must have slipped when I separated the changes into different commits. Can you make a pull request with the fix or should I do it?

We renamed all commandConfiguration to command for uniformity with ros_controllers. It was discussed in #28 and implemented in #30

nrontsis, 2015-06-19:
Thank you for the quick reply :+1: 

Sure, I will make the pull request.

marcoesposito1988, 2015-06-19:
Thank you for the contribution! I'm glad to see the community growing ;)

Please tell us if we can help you getting started or if you have ideas for improvements (especially for a new user, e.g. documentation).

carlosjoserg, 2015-06-19:
Thanks for reporting @nrontsis, actually I meant changing only the topic name, but Marco changed also the function name, I didn't see that.

carlosjoserg, 2015-06-19:
But I see I also proposed to change the function name, sorry.

Again, travis will be of great help here... I'll try to figure out if there is some sort of a hack to use it with 14.04/Indigo.

PR #34 (2015-06-19): Fixing compilation issue regarding commandConfiguration
Fixes #32
closed 2015-06-19 by carlosjoserg