CentroEPiaggio/kuka-lwr #40: Bring effort interfaces back to work in Gazebo

marcoesposito1988, 2015-06-25:
As can be seen in commit 3376577ed8475516c400ef84960576ac7df027d7, removing spring_effort makes the Gazebo simulation work again with effort controllers.

How can we do this in a clean way? Shall we put a yaml parameter to lwr_hw_sim?

carlosjoserg, 2015-06-25:
I will update this next week as well. The idea is to have the same behavior in both lwr_hw_sim and lwr_hw_real, so the plugin will expose both `PositionJointInterface` and `EffortJointInterface`, emulating strategy 10 and 30.

For `EffortJointInterface` is clear, is what exists today that emulates strategy 30.

For `PositionJointInterface` there are two possible variants based on position commands: 
1. Use a PID within the plugin to compute efforts that are sent to the simulation.
2. Use the [SetPosition()](http://osrf-distributions.s3.amazonaws.com/gazebo/api/4.0.0/classgazebo_1_1physics_1_1Joint.html#a5dd5194c47634dde3daf11344b120c32). 

I prefer 2, even when it is not advised for simulation.

carlosjoserg, 2015-07-27:
I believe this is partially solved by #42

closed 2016-10-10 by carlosjoserg