CentroEPiaggio/kuka-lwr #53: One_task_inverse_kinematics controller

manuelbonilla, 2015-09-03:
Hi Guys,

I just cloned this repo.  
I was trying to use the one_task_inverse_kinematics controller on gazebo. 
Using the commands in the readme  the robot did't move. I took a look at the code and I change the  line 130 from
            // joint_handles_[i].setCommand(q_cmd_(i));
to 
            joint_handles_[i].setCommand(joint_des_states_.q(i));
Now the robot moves.
Am I using wrong the controller or is it really an issue in the code?

best

carlosjoserg, 2015-09-03:
Manuel, 

It is an issue in the code, my fault in [this commit](https://github.com/CentroEPiaggio/kuka-lwr/commit/1b3490bf103053503d33ffb8a0bbf64c1d1c2d4c#diff-a43bdeeaa618d731daae33b03e0459c2R130)

Can you PR ithe change ?

carlosjoserg, 2015-09-03:
Solved in #54

closed 2015-09-03 by carlosjoserg