CentroEPiaggio/kuka-lwr #55: Effort controllers failing when using PID class

manuelbonilla, 2015-10-06:
Hi Guys,
I'm trying to use the one_task_inverse_dynamics_JL controller.
It was crashing always when updating. I saw that it is because it is using KinematicChainControllerBase instead of PIDKinematicChainControllerBase.
I suppose it is the same for all controllers generating torque commands or at least the ones using PID interfaces.
am I wrong?

carlosjoserg, 2015-10-07:
Could be, after several HWiface updates, I haven't test how all controllers behave, only a couple of them.

Did you catch the line in which it is crashing?
Are you trying in simulation or in the real robot?

carlosjoserg, 2016-10-10:
I believe this was solved with 07c297b6c5c371d8f048f9ce440ca4c47f33c722

closed 2016-10-10 by carlosjoserg