CentroEPiaggio/kuka-lwr #56: Realtime Safety

davetcoleman, 2015-10-15:
First, great work on this extensive package that supports many interfaces (FRI, Gazebo, etc)!

Question: how realtime safe is this driver implementation? Are you running this on a realtime kernel? I ask because I notice in the [lwr_hw_fril.hpp](https://github.com/CentroEPiaggio/kuka-lwr/blob/master/lwr_hw/include/lwr_hw/lwr_hw_fril.hpp#L55) file you initialize memory every read, which as I understand it is a bad idea. Did someone just overlook this or is it ok here?

Thanks!

carlosjoserg, 2015-10-16:
Hi @davetcoleman

I think we should thank you first for all your contributions to ROS, I'm glad you are putting an eye here.

Actually, I haven't been able to test the Stanford FRIL yet, I just new that I could (I hope my friend @lepalom can help me with that someday ;), he provided the CMake'd version of FRIL for this package). Here, I just used one of the examples Torsten gives to write that. Certainly that's not a good thing to do, but since I haven't even tried yet, not sure if it is ok there or not. We typically use the KUKA FRI@10ms at our site, and 2ms for the most braves.

No, we are not using a realtime kernel, plain Ubuntu 14.04 downloaded from Canonical. The main motivation of the package was to have something "plug-n-play" like, which for a wide range of applications is enough IMHO. In fact, the FRIL does not require a realtime kernel, since in the end, the low-level control is done in the controller box under a _true_ realtime OS. You just need to be in sync with that, other wise FRI complains.

trhermans, 2016-05-31:
I just came across this repository in looking to setup a realtime safe ROS interface to the LWR 4+ which uses the ros_control interface. The alternative solutions out there seem to all build on OROCOS for handling realtime interaction over FRI, the most recent being this fork: https://github.com/kuka-isir/lwr_hardware

I think you are correct that the memory allocation is not good inside the realtime thread.

@carlosjoserg it's unclear to me from your message are you using the other FRI node based on the sample Kuka code for communicating with the control box? This also seems to have some realtime safety issues with memory allocation inside the realtime loop.

My plan is to try and clean this up a bit to be realtime safe. I welcome any feedback on doing this.

carlosjoserg, 2016-06-06:
Hi @trhermans appologies for the delay in the reply.

Yes, that one from ISIR is pretty complete, but also complex.

About he memory allocation, read this: https://github.com/ros-controls/ros_control/issues/213

Yes, the kukaFRI is the default interface for the real hw. More than memory allocation issues, it has problems with threads. This is pretty much solved nicely in the stanfordFRI. If you wish to work on that, that'd be pretty much appreciated.

Note that, I was suggesting to use the cmake'd version of that from https://github.com/iocroblab/fril

closed 2016-10-10 by carlosjoserg