CentroEPiaggio/kuka-lwr #57: Potential Compatibility with IIWA 7 R800

davetcoleman, 2015-10-15:
Hi, do you have any insights into the difficulty of using this package for a IIWA 7 R800? Do I mostly just need to replace the Fast Research Interface (LWRHWFRIL) with Kuka's new Fast *Robot* Interface, creating a new class that inherits from LWRHW? Of course I'd have to update URDFs and other small things too, bug what do you think are the big obstacles? I'm thinking of using this project for that purpose.

Thanks!

carlosjoserg, 2015-10-16:
I don't really know the new IIWA series interfaces, and what they have to offer, hopefully I will soon. In the meantime...

What I've done here is to have an abstract class, LWRHW, written with ros-control tools* that does nothing, since it is pure virtual. Then, the LWRHWFRIL imlpements the init(), read() and write() functions using the FRIL functions.

The main thing is that the LWRHW has some lwr 4+ related stuff which I don't know if they are available in the IIWA series. For instance, I included some KDL stuff, which is useful in Gazebo to add gravity terms, as in the real robot, and so on.

If _abstractively_ speaking they are the same, then yes, it only will take to implement init(), read() and write() functions using the new FRI functions, with the respective changes in the URDFs files (joint name convention).

---
* Btw, I've only detected the use of rosconsle as a dependency, like [here](https://github.com/ros-controls/ros_control/blob/indigo-devel/hardware_interface/include/hardware_interface/internal/resource_manager.h#L40), if you get rid of those using std commands, the ros-control package can be a generic _ros-independent_ language to describe the functionally of robots.

davetcoleman, 2015-10-16:
Thanks for the thoughts!

Re:ROS dependencies - there are actually a number of packages ros_control uses, such as [roscpp](https://github.com/ros-controls/control_toolbox/blob/indigo-devel/package.xml#L21), dynamic_reconfigure, std_msgs, and all the other joint_state messages found in the ros_controllers.

carlosjoserg, 2015-10-16:
Oh, yes, sorry, I meant ifaces packages only (hwiface, jlimitiface,
transmissioniface), not the full ros-controls project ;)

marcoesposito1988, 2015-11-17:
@davetcoleman: sorry for not telling before, but we just made this public right now: in my lab we have been working on implementing a stack for the iiwa, taking this one as a reference, see https://github.com/SalvoVirga/iiwa_stack

We can use the iiwa with MoveIt natively in any mode (we are going to make it possible to configure the ServoMotion with a ROS service). We implemented a ROS node as a Sunrise application, so the communication happens through standard ros topics and we can do anything we want on the Java side. We never had any performance problem with topics, but we also want to provide FRI communication as a drop-in alternative. I would really value your feedback about it and look forward to contributions, since we want to propose it as a standard package.

davetcoleman, 2015-11-17:
Cool, thanks for the heads up on this!

davetcoleman, 2015-11-18:
... but note I already have written my own package using the FRI interface and just joint position control

gavanderhoorn, 2015-11-18:
yay, more IIWA drivers ;)

davetcoleman, 2015-11-18:
lol

gavanderhoorn, 2015-11-18:
:)

closed 2016-10-10 by carlosjoserg