CentroEPiaggio/kuka-lwr #62: joint_impedance_controller (real robot + simulation)

gpldecha, 2016-01-13:
Hi have been trying out the the joint_impedance controller which uses an effort interface.
In your code you send the following torque:

tau_cmd_(i) = K_(i) * (q_des_(i) - q_msr_(i)) + D_(i)*dotq_msr_.qdot(i) + tau_des_(i) + tau_gravity_(i);

Now the LWR 4 kuka already takes into account gravity compensation. Running this on the real robot 
will lead to bad behaviour (in my case). If I send tau_cmd = 0 I get normal gravity compensation with the robot.

I guess the line above is for when using the controller in gazebo. How do you handle  between the real LWR which does torque control via impedance control and the gazebo simulator ?

Thanks.

gpldecha, 2016-01-14:
My bad I got it working... :)

closed 2016-01-14 by gpldecha

carlosjoserg, 2016-01-14:
Hi @gpldecha... however, you were absolutely right. 

Most of the controllers in that folder were developed when the hwiface was more basic, and we wanted to replicate the impedance behavior use it with any other arm as well. 

Nowadays, withing the repo, that term `tau_gravity_` is not needed anymore, since we rely on the impedance controller within the kuka box. And the gazebo plugin also adds that term as well to minimize the efforts of going from sim to real.