CentroEPiaggio/kuka-lwr #65: FRI Interpolation error switching to gravity compensation controller

miguelprada, 2016-02-29:
Whenever I try to switch to the gravity compensation controller I get a FRI Interpolation error which makes the call to `friStart` (after the stiffness strategy is changed) to fail.

I can switch to gravity compensation manually from the pendant, and enter joint impedance control mode using a custom built test executable, so this should rule out erroneous load data parameters, I guess.

I'm running around in circles trying to figure out what's going on. It must be something stupid but I just cannot find it. Has anyone experience something similar or are you capable of running the gravity compensation controller (or any other EffortJointInterface controller, for that matter) without an issue?

miguelprada, 2016-03-02:
Just a quick heads up.

It seems my issue is related to `friRemote` not being thread-safe. I added some quick and dirty synchronization in [`lwr_hw_fri.hpp`](https://github.com/CentroEPiaggio/kuka-lwr/blob/f4da4dd31c1377364fc05ba8325546b303968204/lwr_hw/include/lwr_hw/lwr_hw_fri.hpp) and it seems to fix my issues.

Not sure why this hasn't happened before while using the position controllers. I'll look into it, and try to come up with a PR if I find a clean way to fix the issue.

carlosjoserg, 2016-03-02:
> It seems my issue is related to friRemote not being thread-safe.

Yes, this is it. Sorry for my late reply.

> Not sure why this hasn't happened before while using the position controllers. 

My guess is that the position controllers are less strict in the communication requirements.

> I added some quick and dirty synchronization to lwr_hw_fri.hpp and it seems to fix my issues.

Sorry to making you waste time on that. I didn't want to do it because I know this is already done in the Stanford FRI. The decision initial went to not consider it cause the version I knew used KUKA FRI underneath. But not it is not the case anymore. Anyway, I'm waiting to have some spare time to work with @lepalom and leave the package with the Stanford FRI support as default.

Thanks for the update.

miguelprada, 2016-03-02:
Ok, I'll go on with my dirty fix, then, and stay tuned for further news.

Thanks for the info!

closed 2016-03-02 by miguelprada