CentroEPiaggio/kuka-lwr #80: unable to compile on ros jade

zainmehdi, 2016-11-27:
Hi,
`error: ‘const struct hardware_interface::ControllerInfo’ has no member named ‘hardware_interface’`

this is the error I get when I try to compile. Can you point something out please .

carlosjoserg, 2016-11-28:
Hi @zainmehdi ,

Yes, that's due to API changes in ros control in Jade+, and these packages are working up to Inidgo, so I'll mark the issue as question and not a  bug.

The update is on the way, but with slow response due to other commitments. In the meantime, you are welcome to work on your fork for use&testing. 

I'm aware of other forks with the upgrade, so you are welcome to work it on your fork, and when it is the time, we can discuss about it. What you need to do is something like [this commit](https://github.com/miguelprada/kuka-lwr/commit/86b28a1942bea34a7cb7e3d210d6c8d4387df48a) (which in fact you can add to your git history via adding that fork to your remotes).

Hope it helps for now

zainmehdi, 2016-11-30:
thnx alot it worked

closed 2016-11-30 by zainmehdi