CentroEPiaggio/kuka-lwr #83: Missing moveit_planning_execution.launch

marcoesposito1988, 2017-01-12:
The file was added in cee59d494bf7afe5eda331fbf9cae00a4b4b792f, but I can't find it any more in master.

Was it devoured by some git merge?

carlosjoserg, 2017-01-16:
Hi @marcoesposito1988 Yep, I see it was out in #18, but I don't think on purpose, too many changes in that PR, feel free to add it again, and my apologies !

marcoesposito1988, 2017-01-16:
I thought so :rofl: no problem, I will make a pull request to put it back

marcoesposito1988, 2017-01-19:
BTW: in the current RViz configuration, the planning request marker size is set to zero. This means, that you can't see the marker to command the robot. This can be EXTREMELY confusing for a newbie just trying out the package (I think I spent half an hour figuring this out myself, including recompiling RViz!)

Would you mind if I add another RViz config file to be loaded by moveit_planning_execution.launch? This way your setup would not change, but new users would be not as confused.

marcoesposito1988, 2017-02-10:
Fixed by #88

closed 2017-02-10 by marcoesposito1988