CentroEPiaggio/kuka-lwr #84: Messages for impedance commands

marcoesposito1988, 2017-01-12:
As suggested by @gavanderhoorn in SalvoVirga/iiwa_stack#42, it would probably be a good idea to work on common messages for impedance commands.

There is a starting point at https://github.com/ros-industrial-consortium/majorana (it was a GSoC but it is already over). Shall we both start a branch, and see where it goes from there? I think getting our hands dirty would be the only way to realize real problems with it.

carlosjoserg, 2017-01-16:
Yes, I'm aware of that project too... sure, it sounds clever to work all together in common messages for impedance commands

Do you know of any robot already using them? Or so far it's been what would be ideal to have?

Do you know what was wrong with https://github.com/RCPRG-ros-pkg/cartesian_trajectory_msgs ?

gavanderhoorn, 2017-01-16:
Perhaps @rkojcev can provide some background.

rkojcev, 2017-01-16:
Hi all,

I have tested the set of messages with the kuka iiwa. The driver that interfaces between ROS and the Sunrise OS was the [grl](https://github.com/ahundt/grl).

The goal of the development was to come up with of generalized set of msgs for Impedance/Force control that are robot agnostic.

It would be great if we get some feedback/improvements and more robots using the cartesian impedance messages. Let me know if you need any more info.