CentroEPiaggio/kuka-lwr #85: Support for clamping to a max joint velocity in joint_impedance_controller

marcoesposito1988, 2017-01-16:
While trying out the new joint_impedance_controller, I noticed that sometimes I get an FRI error because of "excessive commanded joint speed" (in T1).

I think it could be useful to pass a maximum desired joint speed to the controller, to which to clamp the command. This could also be desirable for safety purposes.

Would you be interested in a PR?

carlosjoserg, 2017-01-17:
Unfortunately I don't have access to the hardware right now and I can't test it, but sounds good in general, so it might be good to have it at the driver level `lwr_hw`, what do you think?

marcoesposito1988, 2017-01-17:
At that level? Interesting. Yes, could be practical.

I will start playing with it.