CentroEPiaggio/kuka-lwr #89: Gravity compensation controller not working

marcoesposito1988, 2017-02-10:
Hi guys,

I cannot get the gravity compensation controller to work any more, since the big refactoring. 

I get a "FRI Speed limit" error. 

I am on Kinetic / 16.04, since I do not manage to switch control mode on Indigo / 14.04 any more (separate bug report incoming).

Is it just me?

graiola, 2017-06-23:
Hi @marcoesposito1988 , I am facing the same problem with ubuntu 16.04 and kinetic. Did you find a solution? Thanks.

marcoesposito1988, 2017-06-23:
Hi @graiola,

kind of. I hacked it through because I needed it. 

You can take a look at my [gravity compensation branch](https://github.com/marcoesposito1988/kuka-lwr/tree/gravity_compensation). 

However since we are getting rid in the robot in my lab, I am not going to spend much time trying to make a pull request out of it. 

Ciao
Marco